PickNikRobotics/ros2_robotiq_gripper #3: Failing to compile on Humble - cannot find `serial`

jakobbak, 2022-09-14:
Hello

Hope it is ok asking through an issue here.
I am interested in using this package, but trying to compile the source in a workspace fails with the classic
```
  Could not find a package configuration file provided by "serial" with any
  of the following names:

    serialConfig.cmake
    serial-config.cmake
```
I tried running `$ rosdep install --from-paths src -y --ignore-src` but get a response:
```
$ rosdep install --from-paths src -y --ignore-src
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
robotiq_description: Cannot locate rosdep definition for [serial]
robotiq_driver: Cannot locate rosdep definition for [serial]
```
I tried installing both `ros-humble-serial-driver` and `libserial-dev`, but neither of them seems to provide the requested `serialConfig.cmake` file.
 
May I ask you to point me in the right direction of what package you are using and how to get hold of it?

jakobbak, 2022-09-14:
Immediately after posting this I did the obvious and searched for `serial` within the Picknik repositories and found https://github.com/PickNikRobotics/serial-release. 
May I ask then, if it is indeed one of the various forks of William Woodall and John Harrison's `serial` library?
Perhaps https://github.com/PickNikRobotics/serial-release/tree/release/humble/serial?

danzimmerman, 2022-10-27:
I'm working with [Robostack](https://robostack.github.io/humble.html), where the `ros-humble-serial` release isn't available.

I've found that https://github.com/RFRIEDM-Trimble/serial-ros2 builds with `colcon` and works with this repo.

sea-bass, 2023-01-12:
In our repos, we are using https://github.com/wjwwood/serial, and building it from source in the same workspace as `ros2_robotiq_gripper`.

moriarty, 2023-06-13:
closing see #21

closed 2023-06-13 by moriarty