PickNikRobotics/ros2_robotiq_gripper #5: Launch robotiq_driver without having the gripper connect is causing ros2_control_node to segfault

JafarAbdi, 2022-10-19:
Trying to start the gripper without having it connect shouldn't segfault ros2_control_node, this's the logs I'm getting

```
...
demo-drivers-1  | [launcher_node-3] [ros2_control_node-1] Requested 8 bytes, but only got 0.
demo-drivers-1  | [launcher_node-3] [ros2_control_node-1] Requested 8 bytes, but only got 0.
...
demo-drivers-1  | [launcher_node-3] [ros2_control_node-1] Requested 17 bytes, but only got 0.
demo-drivers-1  | [launcher_node-3] [ERROR] [ros2_control_node-1]: process has died [pid 159, exit code -11, cmd '/opt/ros/humble/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_6hguf5xf --params-file /tmp/launch_params_pz9suw_o --params-file /opt/overlay_ws/install/picknik_ur5e_base_config/share/picknik_ur5e_base_config/config/control/picknik_ur5e.ros2_control.yaml --params-file /tmp/launch_params_ynlfyzb3 --params-file /tmp/launch_params_pfmhz6pk -r /streaming_controller/commands:=/robot_controllers/commands -r /joint_trajectory_controller/joint_trajectory:=/robot_controllers/joint_trajectory'].
```

Which are being printed by https://github.com/PickNikRobotics/ros2_robotiq_gripper/blob/main/robotiq_driver/src/robotiq_gripper_interface.cpp#L205

abake48, 2023-05-11:
@MarqRazz can we close this issue?

closed 2023-05-11 by MarqRazz

Cavalletta98, 2025-05-02:
i'm getting the same exact error. does someone solve it?