PickNikRobotics/ros2_robotiq_gripper #9: Failing to compile on foxy - cannot find serial

shubhamsan, 2023-01-11:
Hello
I am  trying to use this package, but while trying to  compile the source in a workspace fails with the classic
 ```
Could not find a package configuration file provided by "serial" with any
  of the following names:

    serialConfig.cmake
    serial-config.cmake
```
I tried running `$ rosdep install --from-paths src -y --ignore-src` but get a response:

```
$ rosdep install --from-paths src -y --ignore-src
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
robotiq_description: Cannot locate rosdep definition for [serial]
robotiq_driver: Cannot locate rosdep definition for [serial]

```
Can any one help me with this, I also refered https://github.com/PickNikRobotics/ros2_robotiq_gripper/issues/3 but still facing the issue.

sea-bass, 2023-01-12:
In our repos, we are using the `main` branch of https://github.com/wjwwood/serial, and building it from source in the same workspace as `ros2_robotiq_gripper`.

moriarty, 2023-06-13:
closing see #21

closed 2023-06-13 by moriarty