PickNikRobotics/ros2_robotiq_gripper #21: Serial (Dependency Missing) [Meta Issue]

moriarty, 2023-06-13:
Creating one top issue to track this and  Closing #3 & closing #9 Which were duplicates for Foxy and Humble.

Unfortunately Foxy reaches EOL next week, so I do not expect serial will not be available in time for the last sync.

The underlying Issue is the same, the dependency  https://github.com/wjwwood/serial has not been released to ROS 2.
See https://github.com/wjwwood/serial/issues/204 and https://github.com/wjwwood/serial/issues/209 and https://github.com/wjwwood/serial/pull/209#issuecomment-1538485492

@wjwwood has mentioned https://github.com/wjwwood/serial/pull/133#issuecomment-357464182 and https://github.com/wjwwood/serial/pull/231#issuecomment-859110462 that he would like to move serial to a pure CMake package, there is a branch here https://github.com/cottsay/serial/commit/4d5be00015bb1aad1017a064408e884d4af388db which does this. I've tried to collect these issues here https://github.com/wjwwood/serial/issues/283

There was a request from @SteveMacenski https://github.com/wjwwood/serial/issues/204#issuecomment-686776892 to move the serial repo to it's own location, which I've opened as an issue here https://github.com/wjwwood/serial/issues/284

The Robotiq gripper isn't the only package which has been using serial, @tonybaltovski pointed out here https://github.com/wjwwood/serial/pull/209#issuecomment-1538507835 that they're working on porting some other packages to ROS 2

There are several branches and forks which can be found from going through the above issues.
Currently we've been using this package with this branch, on this fork https://github.com/tylerjw/serial/tree/ros2 which was started off of this open PR https://github.com/wjwwood/serial/pull/209 which is pretty close to this https://github.com/RoverRobotics-forks/serial-ros2

There alternative mentioned here: https://github.com/wjwwood/serial/issues/204#issuecomment-1266424111 is not a drop in replacement, it uses boost asio which throws different exceptions.

Another alternative is to use the debian which is released as `libcxx-serial-dev` which comes from this PR https://github.com/wjwwood/serial/pull/231 on this fork & branch https://github.com/leamas/serial/tree/pr

But all the different branches and forks are slightly diverging and we'll need to inspect the commit history and test.
We know the branch we've been using is pretty well tested.

Ryanf55, 2023-06-16:
Let me know how I can help. 

The asio version worked well in a few vehicles I used. Considering that the STL's upcoming support of networking is modeled after asio, I recommend trying to go that route. Many people start with `polling` and blocking reads, but the async style support is more performant and I think is more familiar to ROS devs that use a pub/sub model.

moriarty, 2023-06-16:
> Let me know how I can help.
>
> The asio version worked well in a few vehicles I used. Considering that the STL's upcoming support of networking is modeled after asio, I recommend trying to go that [...]

@Ryanf55 did you use this version: https://github.com/ros-drivers/transport_drivers/tree/main/serial_driver

I did start going down that path, I'll see if I still have the branch around locally.

As I mentioned above, we've been using this branch: https://github.com/tylerjw/serial/tree/ros2 and know works and hadn't planned all the extra testing if we do swap to serial... I have a robotiq gripper on an arm and I need to ask a colleague with the serial adaptor to run hardware tests for me... but it does seem that we should be moving away from the https://github.com/wjwwood/serial dependency if https://github.com/wjwwood/serial/issues/284 is not addressed or if https://github.com/wjwwood/serial/pull/285 merged

Ryanf55, 2023-06-16:
> 

I've used both, deployed both libraries in the field, ran at 115k and 230k baud rates. Both worked. I was chasing down some communication issues with the other device, which turned out to be a fault in the embedded device. 

I know this one works: https://github.com/iwelch82/serial-ros2

It's interesting, I already add PIC, the commit here: https://github.com/tylerjw/serial/commit/d8d160678aa0b31cdf467c052b954fa287cc6cdf
is the same idea as what I did here: 
https://github.com/iwelch82/serial-ros2/commit/8b6c14dbdee29ba8fcaa6473c702376a099adc75

This just shows the need to consolidate efforts and stop forking basic drivers like serial.

Ryanf55, 2023-06-17:
Using asio might be a bit tricky. With asio taking on the approach to use C++20 for the features like coroutine in their [1.17.0](https://think-async.com/Asio/asio-1.28.0/doc/asio/history.html#asio.history.asio_1_17_0) release, which [seem really great to use](https://www.youtube.com/watch?v=icgnqFM-aY4&t=2954s), since ROS 2 humble is on C++17, I don't know if the ROS organization would allow a bump to C++20 minimum for the serial driver. Perhaps we could have a chat, next week, to determine an approach? I'm on the ROS discord as `RyanF`.

atomoclast, 2024-01-26:
Hi all,

I see this is still an open issue. Is there a proposed workaround for getting this repo to build for use in ROS2? I'm definitely hitting the wall of build failures since there's no serial package that seems valid?

karanchahal, 2024-08-06:
I see this issue as well

itzikcyberbee, 2024-09-06:
i also suffer from this, still no fix?