PickNikRobotics/ros2_robotiq_gripper #43: Gripper velocity and force can only be configured by editing the URDF

pac48, 2024-01-26:
The gripper closing speed and closing force should be configurable at runtime.  The `robotiq_activation_controller` package does not provide a mechanism to set the speed and force. The problem can be fixed by adding ROS parameters to `robotiq_activation_controller` and slightly modifying the `hardware_interface` to export interfaces to control force and velocity.

pac48, 2024-02-05:
Blocked by https://github.com/PickNikRobotics/moveit_studio/issues/6256

nbbrooks, 2024-06-12:
I noticed https://github.com/PickNikRobotics/moveit_studio/issues/6256 is closed (ros2 control maintainers fixed it another way) - does that mean this is unblocked?

pac48, 2024-06-12:
> I noticed [PickNikRobotics/moveit_studio#6256](https://github.com/PickNikRobotics/moveit_studio/issues/6256) is closed (ros2 control maintainers fixed it another way) - does that mean this is unblocked?

I think it is unblocked now, but still waiting on this PR to go in https://github.com/ros-controls/ros2_controllers/pull/1002

nbbrooks, 2024-07-11:
It's merged!!!

closed 2024-07-25 by pac48