PickNikRobotics/ros2_robotiq_gripper #49: robotiq_85_2f_gripper appears broken in Gazebo - no mimic_joints

ProfJust, 2024-05-24:
The parallel kinematic with mimic_joints used does not work in ROS2-Humble with Gazebo-Classic. The Gripper is broken.

This the same in this issue https://github.com/ros-industrial/robotiq/issues/150.

[image]

Unfortunately there is no [roboticsgroup_gazebo_plugin] (https://github.com/roboticsgroup/roboticsgroup_gazebo_plugins) available for ROS2.

How can I get this gripper working with ROS2?

theanhnguyen3112, 2024-09-25:
Hello, have you fixed it yet?

ProfJust, 2024-09-25:
Unfortunately not.   I tried to get it working with the new Gazebo (Ignition) at ROS2-Jazzy, but there I found the same problems,
As a workaround I fixed the Joints, so the gripper looks OK but cannot be opened/closed.

nbbrooks, 2024-12-27:
For Gazebo Ignition issues involved mimic joints (used by Robotiq 2F and other devices), you should file an issue to https://github.com/ros-controls/gz_ros2_control asking which versions of gz-sim are expected to support mimic joints.

closed 2024-12-27 by nbbrooks