PickNikRobotics/ros2_robotiq_gripper #52: How to use the repo?

psemalah, 2024-06-29:
Really sorry for the stupid question, how to use the repo for binary install? Many Thanks.

nbbrooks, 2024-12-27:
Closing due to inactivity to response.

closed 2024-12-27 by nbbrooks

yijionglin, 2025-01-16:
Hi @abuibaid ,

Thanks for the answer, it really helps someone like me who is new to ros2.

I followed your answer and after running all the commands, I tried to run "ros2 launch robotiq_driver"

```
ros2 launch robotiq_driver 
file 'None' was not found in the share directory of package 'robotiq_driver' which is at '/home/bourne/ros2/robotiq_workspace/install/robotiq_driver/share/robotiq_driver'
```

Could you please provide any suggestions on this?

Many thanks,
Bourne

nbbrooks, 2025-01-16:
It is reporting that error because you did not specify a launch file in the `robotiq_driver` package - my guess is Python is then making it default to finding a launch file named "None"

Try running `ros2 launch robotiq_description robotiq_control.launch.py`

iamshakeelsindhu, 2025-05-13:
May I know, do you have any recommendations, how to use this repo for running IsaacSim simulation of robotiq_2F_85 (or maybe another one). 

I can use this in rviz.
For IsaacSim, I am using IsaacSim Assets (to load robotiq_2f_85) and then try to run it, but of course, it asks for the real gripper. Additionally, I created a launch file, but the gripper just explodes. (And that's the problem with my launch file, that it only controls a few joints, the rest just exlodes, PhysX error for the joints.)

Looking forward to the recommendations.

Thank You