PickNikRobotics/ros2_robotiq_gripper #53: Integrate this driver with MoveIt Task constructor

karanchahal, 2024-07-01:
Hello,

Thanks for this package, how would we use this driver and it's low level fuctions like.
```
driver->set_gripper_position(0xFF);
    while (driver->gripper_is_moving())
    {
      std::this_thread::sleep_for(std::chrono::milliseconds(500));
    }

    std::cout << "Opening the gripper..." << std::endl;
    driver->set_gripper_position(0x00);
    while (driver->gripper_is_moving())
    {
      std::this_thread::sleep_for(std::chrono::milliseconds(500));
    }

    std::cout << "Half closing the gripper..." << std::endl;
    driver->set_gripper_position(0x80);
```

like `set_gripper_position` function in the driver, through the move it task constructor interface of 
```
auto move_to = std::make_unique<mtc::stages::MoveTo>("move to", std::make_shared<mtc::solvers::JointInterpolationPlanner>());
    move_to->setGroup(arm_group_name);
    move_to->setGoal({{"gripper_left": 0.75, {"gripper_right": 0.75}}); # this is just doing joint t=interpolation
    task.add(std::move(move_to));
```
joint interpolation is not a thing in the real robotiq driver, so I guess how would move it task constructor plumb a position control + velocity / force command ?

ashwinvkNV, 2024-07-13:
I don't think you can control the gripper_left and gripper_right independently because the gripper itself cannot control both these joints independently.
You can only send a "POSITION REQUEST". Refer page 42 of [this manual](https://assets.robotiq.com/website-assets/support_documents/document/2F-85_2F-140_Instruction_Manual_e-Series_PDF_20190206.pdf)

Instead you can control the "finger_joint" mentioned in the urdf through task constructor as shown below.

```
auto move_to_close = std::make_unique<mtc::stages::MoveTo>("move to", std::make_shared<mtc::solvers::JointInterpolationPlanner>());
    move_to_close->setGroup(arm_group_name);
    move_to_close->setGoal({{"finger_joint", 0.623}});
```

In order to this you also have to specify a srdf file. I have attached the urdf, srdf, launch and task_constructor files for reference. This was tested on a real life 2F-140. And I was able to open and close the gripper.

 
[robotiq_moveit_task_constructor.zip](https://github.com/user-attachments/files/16203688/robotiq_moveit_task_constructor.zip)

JafarAbdi, 2025-03-05:
Thanks @ashwinvkNV for your answer. Closing this issue!

closed 2025-03-05 by JafarAbdi