PickNikRobotics/ros2_robotiq_gripper #60: Building issues with robotiq_driver

aalmrad, 2024-07-22:
Hello,
My name is Abed and I have been working with the `ros2_robotiq gripper` repository provided by your team. First, I would like to express my gratitude for the excellent work you have done with this package.

I have faced some issues while building the `robotiq_driver` package with colcon.  After some investigation and testing, I was able to implement a fix that resolved the problem. This required the application of the following patch

```
diff --git a/robotiq_driver/tests/test_robotiq_gripper_hardware_interface.cpp b/robotiq_driver/tests/test_robotiq_gripper_hardware_interface.cpp
index b187a80..d09b8de 100644
--- a/robotiq_driver/tests/test_robotiq_gripper_hardware_interface.cpp
+++ b/robotiq_driver/tests/test_robotiq_gripper_hardware_interface.cpp
@@ -76,7 +76,7 @@ TEST(TestRobotiqGripperHardwareInterface, load_urdf)
   rclcpp::Node node{ "test_robotiq_gripper_hardware_interface" };
 
   // Initialize the resource manager
-  hardware_interface::ResourceManager rm(urdf, node.get_node_clock_interface(), node.get_node_logging_interface());
+  hardware_interface::ResourceManager rm(urdf, true, false);
 
   // Check interfaces
   EXPECT_EQ(1u, rm.system_components_size());
```
   
   It seems that the issue was related to an outdated method's signature.

sea-bass, 2024-08-01:
Not outdated, just that it changed recently in rolling.

If you grab the humble branch of this repo, it should contain your suggested interface.

aalmrad, 2024-08-08:
Thank you @sea-bass for pointing that out! The humble branch worked perfectly with no build issues

closed 2024-12-27 by nbbrooks