PickNikRobotics/ros2_robotiq_gripper #63: Better Error Messge when Gripper is not connected

Abishalini, 2024-08-15:
During bringup, the RobotiqGripperHardwareInterface throws the following error if the gripper is not connected. 
We need to get a better error message about the failure to connect to the gripper.

```
drivers-1       | [ros2_control_node-4] [INFO] [1723749978.912978825] [resource_manager]: 'configure' hardware 'RobotiqGripperHardwareInterface' 
drivers-1       | [ros2_control_node-4] [ERROR] [1723749978.913051693] [RobotiqGripperHardwareInterface]: Cannot configure the Robotiq gripper: IO Exception (2): No such file or directory, file /home/demo/user_ws/src/moveit_studio_ur_ws/src/external_dependencies/serial/src/impl/unix.cc, line 151.
```

Mokchanic, 2024-08-26:
Hi did you connect the gripper to the control box?

Abishalini, 2024-08-27:
Hi @Mokchanic 

The 2f-85 gripper was connected through USB to a computer. Before I figured out to set the `com_port`field in the xacro, I was stumped by the error message mentioned above. There should be a better message to indicate gripper connection is not detected.

Mokchanic, 2024-08-29:
Hi @Abishalini 
Does the computer not detect the port connection or is it designated ttyUSB0?
Check out udevadm.

```bash
sudo udevadm monitor
```

Find your ttyUSB number while connecting or removing the port.
[image]

By the way, are you using 'ros humble'?
Did you run the .repos file?

abuibaid, 2024-09-19:
Hi @Abishalini, 

were you able to fix this issue?

Aruen24, 2024-11-19:
> During bringup, the RobotiqGripperHardwareInterface throws the following error if the gripper is not connected. We need to get a better error message about the failure to connect to the gripper. [...]

@Abishalini Hi, Have you fixed the problem
I also report the error. 
Ros2 humble  ubuntu 22.04.   
the port connection /tmp/ttyUR

[ERROR] [1732020153.874565794] [RobotiqGripperHardwareInterface]: Cannot configure the Robotiq gripper: IO Exception (2): No such file or directory, file /src/planning_control-1119/serial-ros2/src/impl/unix.cc, line 152.