PickNikRobotics/ros2_robotiq_gripper #64: No "serial" package?

katallen405, 2024-09-12:
When I colcon build from the Humble branch, it does not find a "serial" package, and rosdep does not find "serial" to install either.
There is a serial-driver -- is that what is wanted, or is there some other serial package that should be installed that is not a ROS standard package?

abuibaid, 2024-09-19:
Hi @katallen405,

if I'm not mistaken, I think you need run these two commands before running colcon build, where the second command will import the serial-package

```
git clone -b humble https://github.com/PickNikRobotics/ros2_robotiq_gripper.git src/<name-of-the-folder>
vcs import src --skip-existing --input src/<name-of-the-folder>/ros2_robotiq_gripper-not-released.humble.repos
```

nbbrooks, 2024-12-27:
@katallen405 can you confirm if this resolves the issue?

JafarAbdi, 2025-03-05:
Closing as @abuibaid response should address this.

closed 2025-03-05 by JafarAbdi