PickNikRobotics/ros2_robotiq_gripper #72: Using `package://` prefixes breaks Gazebo model loading

sea-bass, 2024-12-19:
https://github.com/PickNikRobotics/ros2_robotiq_gripper/pull/50 was merged somewhat recently, which I just learned in testing `ros2_kortex` repo that it breaks the Gazebo loading.

If I revert this PR, the gripper loads fine in Gazebo.

traversaro, 2024-12-19:
Gazebo supports package: URIs , but what it does it just converts them to model: , see https://github.com/gazebosim/sdformat/blob/69c82c00fae5102f360b299ff6f396bab126ced3/src/parser_urdf.cc#L2412 . So I guess you need to append <install_prefix>/share to GAZEBO_MODEL_PATH (for Gazebo Classic) or  GZ_SIM_RESOURCE_PATH (for gz-sim)

sea-bass, 2024-12-19:
Thanks @traversaro!

I think I may have added the wrong path or something, as I did try modifying `GZ_SIM_RESOURCE_PATH` both in my terminal and in the launch file. I'll try again.

To confirm, if the install path looks like:

```
/home/sebastian/my_workspace/install/robotiq_description/share/robotiq_description/meshes/...
```

What exactly should be the path added to the env variable? Is it `get_ament_package_prefix("robotiq_description")` or am I off by a few directories?

traversaro, 2024-12-19:
If I am not wrong, you should set `GZ_SIM_RESOURCE_PATH` to `/home/sebastian/my_workspace/install/robotiq_description/share`, but I am not 100% sure.

sea-bass, 2024-12-19:
Indeed, I was off by one (as are all the best engineering problems) -- you are right.

Thanks, @traversaro!

[image]

closed 2024-12-19 by sea-bass