PickNikRobotics/ros2_robotiq_gripper #73: Loader for controller 'robotiq_activation_controller' (type 'robotiq_controllers/RobotiqActivationController') not found.

tajish, 2025-01-23:
When I try to configure and action the robotiqActivationController using gazebo, it throws an error saying:

Loader for controller 'robotiq_activation_controller' (type 'robotiq_controllers/RobotiqActivationController') not found.

And from the controller launch file it shows:

Failed loading controller robotiq_activation_controller

this is my controller.yaml file:

controller_manager:
  ros__parameters:
    update_rate: 10 # Hz

    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

    joint_trajectory_controller:
      type: joint_trajectory_controller/JointTrajectoryController
    
    robotiq_gripper_controller:
      type: position_controllers/GripperActionController

    robotiq_activation_controller:
      type: robotiq_controllers/RobotiqActivationController

joint_trajectory_controller:
  ros__parameters:
    joints:
      - shoulder_pan_joint
      - shoulder_lift_joint
      - elbow_joint
      - wrist_1_joint
      - wrist_2_joint
      - wrist_3_joint
    command_interfaces:
      - position
    state_interfaces:
      - position
    state_publish_rate: 100.0
    action_monitor_rate: 20.0
    allow_partial_joints_goal: false
    constraints:
      stopped_velocity_tolerance: 0.2
      goal_time: 0.0
      shoulder_pan_joint: { trajectory: 0.2, goal: 0.1 }
      shoulder_lift_joint: { trajectory: 0.2, goal: 0.1 }
      elbow_joint: { trajectory: 0.2, goal: 0.1 }
      wrist_1_joint: { trajectory: 0.2, goal: 0.1 }
      wrist_2_joint: { trajectory: 0.2, goal: 0.1 }
      wrist_3_joint: { trajectory: 0.2, goal: 0.1 }

robotiq_gripper_controller:
  ros__parameters:
    default: true
    joint: robotiq_85_left_knuckle_joint
    use_effort_interface: true
    use_speed_interface: true

robotiq_activation_controller:
  ros__parameters:
    default: true

And this is the launch file I am using:

from launch import LaunchDescription
from launch_ros.actions import Node
import os
from launch_ros.parameter_descriptions import ParameterValue
from launch.substitutions import Command
from ament_index_python.packages import get_package_share_directory
import launch_ros

def generate_launch_description():

    robot_description = ParameterValue(
        Command(
            [
                "xacro ",
                os.path.join(get_package_share_directory("ur5_robot_description"),"urdf","ur5_gripper_main.xacro")
            ]
        ),
        value_type=str
    )

    robot_state_publisher_node = Node(
        package="robot_state_publisher",
        executable="robot_state_publisher",
        parameters=[{
            "robot_description": robot_description,
            "use_sim_time": True
        }]
    )

    joint_state_broadcaster_spawner = Node(
        package="controller_manager",
        executable="spawner",
        arguments=[
            "joint_state_broadcaster",
            "--controller-manager",
            "/controller_manager"
        ]
    )

    joint_trajectory_controller_spawner = Node(
        package="controller_manager",
        executable="spawner",
        arguments=[
            "joint_trajectory_controller",
            "--controller-manager",
            "/controller_manager"
        ]
    )

    robotiq_gripper_controller_spawner = Node(
        package="controller_manager",
        executable="spawner",
        arguments=[
            "robotiq_gripper_controller",
            "--controller-manager",
            "/controller_manager"
        ]
    )

    robotiq_activation_controller_spawner = Node(
        package="controller_manager",
        executable="spawner",
        arguments=["robotiq_activation_controller", "-c", "/controller_manager"],
    )

    return LaunchDescription([
        robot_state_publisher_node,
        joint_state_broadcaster_spawner,
        joint_trajectory_controller_spawner,
        robotiq_gripper_controller_spawner,
        robotiq_activation_controller_spawner
    ])

Please help me out in dealing with it.

JafarAbdi, 2025-03-05:
Could you please share the logs? I'll close it for now, please feel free to re-opened it if you are still running into this issue.

closed 2025-03-05 by JafarAbdi

fzoric8, 2025-05-20:
@JafarAbdi, @Cavalletta98 

I'm having the same issue.