PickNikRobotics/ros2_robotiq_gripper #87: Build issue in ROS2 Jazzy

brolinA, 2025-04-17:
I was able to build the [serial-ros2](https://github.com/RoverRobotics-forks/serial-ros2) package by putting it within the colcon workspace. The serial package builds successfully but the `robotiq_driver` fails to build with the following error message. 

```
--- stderr: robotiq_driver                                                                                             
/usr/bin/ld: /home/ur5-akzente/ur_ws/install/serial/lib/libserial.a(unix.cc.o): warning: relocation against `_ZTVN6serial15SerialExceptionE' in read-only section `.text._ZN6serial15SerialExceptionD2Ev[_ZN6serial15SerialExceptionD5Ev]'
/usr/bin/ld: /home/ur5-akzente/ur_ws/install/serial/lib/libserial.a(serial.cc.o): relocation R_X86_64_PC32 against symbol `_ZTVN6serial6SerialE' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/robotiq_driver.dir/build.make:392: librobotiq_driver.so] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:229: CMakeFiles/robotiq_driver.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< robotiq_driver [15.3s, exited with code 2]
```

could you please help with the issue?

s4lmon, 2025-04-18:
Getting a similar issue with serial:
````
--- stderr: robotiq_driver                                                                                                                      
CMake Error at CMakeLists.txt:18 (find_package):
  By not providing "Findserial.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "serial", but
  CMake did not find one.

  Could not find a package configuration file provided by "serial" with any
  of the following names:

    serialConfig.cmake
    serial-config.cmake

  Add the installation prefix of "serial" to CMAKE_PREFIX_PATH or set
  "serial_DIR" to a directory containing one of the above files.  If "serial"
  provides a separate development package or SDK, be sure it has been
  installed.

---
```

brolinA, 2025-04-18:
> Getting a similar issue with serial:
>
> ```
> --- stderr: robotiq_driver
> CMake Error [...]

This is likely because you didn't install the serial-ros2. Please follow the link in my original question to install it. Then there shouldn't be an issue.

closed 2025-04-24 by brolinA